acvigue/esphome-levoit-air-purifier #4: Unexpected Behavior of Homeassistant Components

haydonryan, 2024-05-09:
Me again.  I apologize this is going to be a bit of a word soup.

My ESPhome is hooked up to homeassistant with my other devices.  Overall working really really well. The integration creates 1 device and 10 entities. 

The fan entity it isn't doing what I think it should be doing (and also not working per the old VEsync integration).

fan.office_purifier can be set to 0, 33%,66% and 100%
switch.office_purifier_master_power (on or off).
switch.office_purifier_master_power (Default, Quiet and Efficient)

The VEsync integration worked that 0 for fan.office_purifier (fan) was the fan not spinning. There was a switch for auto mode (on / off) and sleep mode (on / off).  

With this integration if I set the fan to zero,  this changes the select.office_purifier_fan_mode to go from manual to  "auto" mode, which spins at a very slow speed.    At the moment I'm overcoming this by also turning switch.office_purifier_master_power to off.

select.office_purifier_auto_mode stays on "default", but has options for Quiet and Efficient, but neither of these makes sense given  that auto it's either on auto or not.

What I'd love to have happen is that setting the fan to 0, keeps the fan on manual, stops the fan from spinning (this would align it to the ceiling fan, and old vesync integration).  

Currently if i turn on the purifier - I'll set the fan to 33% and that will turn it on, but to turn it off I have to use the master power switch.

haydonryan, 2024-05-13:
We definitely need to adjust this - automatons fail to work properly in home assistant + homekit with ios..

EG upstairs right now my purifier is on 33% however  master power is "off"

For Automations:
Setting fan to zero% results in turning master power on (if it's showing off).
Turning the master power off, resutls in the fan still showing 33% speed even though it's not running.

haydonryan, 2024-05-21:
Sorry i know more information -  When creating scenes in homeassistant you can only add per device, so one entity can't rely on another entity - eg can't set fan to 0% AND turn off master power - ends in a race condition.

SeveranExp, 2024-11-01:
Oh, hey yep, you're seeing similar behavior to me.

haydonryan, 2024-12-10:
I can invest some time to try change the logic to have an auto switch and have the fan go to zero. @acvigue wre you ok with this change?

Grafton1046, 2025-01-07:
I added a template switch helper in home assistant to cover for the "auto" mod showing as the fan was off:
[image]

haydonryan, 2025-01-15:
@Grafton1046 So did this get added to the room? I'm a little confused what you're using this switch for?

Grafton1046, 2025-01-15:
I misread your inital issue, I just wanted a way to show via a switch it was in auto mode, to up the wife approval factor.

mmathau, 2025-07-02:
I am seeing the same issues as @haydonryan in his original report.
At the moment I am doing something like this for a night routine that sets the core300s to sleep/quiet mode for the bedroom.
```yaml
alias: Set Sleep Mode on Air Purifier
if:
  - condition: state
    entity_id: fan.esphome_levoit300s_bedroom_air_purifier
    state: "on"
then:
  - action: switch.turn_off
    metadata: {}
    data: {}
    target:
      entity_id: switch.esphome_levoit300s_bedroom_display_on
  - action: select.select_option
    metadata: {}
    data:
      option: Sleep
    target:
      entity_id: select.esphome_levoit300s_bedroom_fan_mode
  - action: select.select_option
    metadata: {}
    data:
      option: Sleep
    target:
      entity_id: select.esphome_levoit300s_bedroom_auto_mode
    enabled: false
```
With this the `fan.esphome_levoit300s_bedroom_air_purifier` entity shows as `OFF` in homeassistant, which technically is true since levoit states in the manual "Sleep Mode operates quietly by using a fan speed lower than fan speed 1." and the `switch.esphome_levoit300s_bedroom_power` stays `ON`