acvigue/esphome-levoit-air-purifier #5: What would be needed to add support for other Levoit air purifiers?

andreklug, 2024-06-06:
Hi,

I've got the Levoit Vital 100S and 200S devices. I am very keen to use them with esphome.
I am more than happy to disassemble them and do some soldering. 

Could you please outline the steps that I would need to do in order to provide all information you'd need to add support for these devices as well?

Thank you!

acvigue, 2024-06-09:
Hi @andreklug ! Please upload high resolution photos of the control board. It should have an ESP32 module or something similar.

andreklug, 2024-06-09:
Here you go: 

Levoit Vital 100S, will disassemble the 200 next.

[image]
[image]

acvigue, 2024-06-09:
[image]

This board please

andreklug, 2024-06-10:
oh that was stupid. Anyway, I am awake now:

[image]
[image]
[image]

andreklug, 2024-06-10:
a bit further information:

(to get this this into download mode, IO0 needs to be connected to ground when powering up / connecting to UART adapter)

This is how the device identifies itself:

```
esptool.py v4.7.0
Serial port /dev/cu.usbserial-0001
Connecting....
Detecting chip type... ESP32-C3
Chip is ESP32-C3 (QFN32) (revision v0.4)
Features: WiFi, BLE, Embedded Flash 4MB (XMC)
Crystal is 40MHz
```

I would now try your code as is to see if it already works. But as I have only been using esphome to flash precompiled stuff, I have no idea where to start compiling it. Would you mind giving me a short explaination?

Thank you!

acvigue, 2024-06-10:
Sample config is given in repo - it will pull this code and compile for you

andreklug, 2024-06-11:
Thank you. I created and adpated the config file. Deleted the flash and wrote it to the board. Setting the board to a generic ESP32-C3 compiles just fine, however the serial monitor shows "rst: 0x10 (RTCWDT_RTC_RST),boot:0xd (SPI_FLASH_BOOT) and then many lines of "invalid header: 0xffffffff".

```
I tried another C3 board and git this config error:

INFO ESPHome 2024.5.5
INFO Reading configuration /config/esphome/luftfilter-buro.yaml...
Failed config

uart: [source /config/esphome/luftfilter-buro.yaml:38]
  - 
    This pin cannot be used on ESP32-C3s and is already used by the SPI/PSRAM interface (function: SPIQ).
    tx_pin: GPIO17
    
    This pin cannot be used on ESP32-C3s and is already used by the SPI/PSRAM interface (function: SPID).
    rx_pin: GPIO16
    baud_rate: 115200

```

I guess the next step would be to identify the correct GPIO header. Any hint how to do this?

acvigue, 2024-06-11:
Please upload full redacted with config

andreklug, 2024-06-11:
In this very second I am trying the more generic "esp32dev" board. Here's the config:

```
esphome:
  name: "luftreiniger-buero"
  friendly_name: "Luftreiniger Büro"

esp32:
  board: esp32dev
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_FREERTOS_UNICORE: y

external_components:
  - source: github://acvigue/esphome-levoit-purifier
    components: [levoit]

logger:
  baud_rate: 115200

# Enable Home Assistant API
api:
  encryption:
    key: "REDACTED"

ota:
  password: "REDACTED"

wifi:
  networks:
    - ssid: !secret wifi_ssid_btst
      password: !secret wifi_password_btst

  ap:
    ssid: "LR-Buero Fallback"
    password: !secret ap_password

captive_portal:

uart:
  tx_pin: GPIO21
  rx_pin: GPIO20
  baud_rate: 115200

levoit:
  id: levoit1

fan:
  - platform: levoit
    name: Levoit Purifier

sensor:
  # Uptime sensor.
  - platform: uptime
    name: Uptime

  # WiFi Signal sensor.
  - platform: wifi_signal
    name: WiFi Signal
    update_interval: 60s

  # PM2.5 sensor
  - platform: levoit
    pm_2_5:
      name: PM2.5
    air_quality:
      name: Air Quality Index

switch:
  - platform: levoit
    display_lock:
      name: "Display Lock"
    display_on:
      name: "Display On"
    power:
      name: "Master Power"

select:
  - platform: levoit
    fan_mode:
      name: "Fan Mode"
    auto_mode:
      name: "Auto Mode"
```

andreklug, 2024-06-11:
esp32dev as board results in the "invalid header" error.

acvigue, 2024-06-11:
It should be the same board as the one I used c3 solo

andreklug, 2024-06-11:
no luck... 

[image]

andreklug, 2024-06-11:
This is what I can select for C3:

[image]

andreklug, 2024-06-11:
So I tried esp32-c3-devkitc-02 (not found), devkitm1, m1i-kit. After a successful upload, both are returning this:

```
rst:0x10 (RTCWDT_RTC_RST),boot:0xd (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:QIO, clock div:1
load:0x3fcd5810,len:0x1684
ets_loader.c 78 
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
```

acvigue, 2024-06-11:
Do you have a dump of the original firmware

andreklug, 2024-06-11:
yes, you want me to share it?

acvigue, 2024-06-11:
yes

andreklug, 2024-06-11:
[levoit.bin.zip](https://github.com/user-attachments/files/15794177/levoit.bin.zip)

acvigue, 2024-06-14:
Please also upload the binary you are attempting to flash to the device. Since it contains encryption keys & such, you may email it to me if you'd like...

acvigue, 2024-06-14:
```bash
~ ❯ esptool.py image_info Downloads/levoit.bin --version 2
esptool.py v4.7.0
File size: 4194304 (bytes)
Detected image type: ESP32-C3

ESP32-C3 image header
=====================
Image version: 1
Entry point: 0x403ce000
Segments: 3
Flash size: 4MB
Flash freq: 80m
Flash mode: DIO

ESP32-C3 extended image header
==============================
WP pin: 0xee (disabled)
Flash pins drive settings: clk_drv: 0x0, q_drv: 0x0, d_drv: 0x0, cs0_drv: 0x0, hd_drv: 0x0, wp_drv: 0x0
Chip ID: 5 (ESP32-C3)
Minimal chip revision: v0.0, (legacy min_rev = 3)
Maximal chip revision: v0.0

Segments information
====================
Segment   Length   Load addr   File offs  Memory types
-------  -------  ----------  ----------  ------------
      1  0x011c4  0x3fcd6100  0x00000018  DRAM, BYTE_ACCESSIBLE
      2  0x008a8  0x403ce000  0x000011e4  IRAM
      3  0x02ca0  0x403d0000  0x00001a94  IRAM

ESP32-C3 image footer
=====================
Checksum: 0x5d (valid)
Validation hash: db2ef7e43fbc89495281279dbd0ef193adca7a40ab28a34b67d3839f6113d226 (valid)
```

SeveranExp, 2024-11-01:
@acvigue hey friend, are you still willing to work on this firmware?

moimeme81, 2024-11-06:
> @acvigue hey friend, are you still willing to work on this firmware?
I was wondering the same. @andreklug anything new on your side?

brcommander, 2025-01-04:
would be interested too

andreklug, 2025-01-04:
Sorry for the late reply. I decided to go with the already supported 300 and 400 models for my home, as some other stuff took too much of my time.

brcommander, 2025-01-04:
Oh I see...so I will have to go on myself. I managed to flash a custom esphome firmware and I see it in Home Assistant but neither the controls nor the sensors (pm 2.5...) are functional...

brcommander, 2025-01-05:
so I got a kind of esphome firmware up, getting logs via wifi from esphome.
[19:57:35][C][levoit:066]: Levoit!
[19:57:39][VV][scheduler:225]: Running interval 'heartbeat' with interval=15000 last_execution=854217 (now=869233)
[19:57:39][V][levoit:019]: Sending heartbeat
[19:57:39][V][levoit:209]: Sending CMD (013140): A5.22.84.04.00.3E.01.31.40.00 (10)
[19:57:39][W][levoit:233]: ACK response timeout, retry #1
[19:57:39][V][levoit:209]: Sending CMD (013140): A5.22.85.04.00.3D.01.31.40.00 (10)
[19:57:39][W][levoit:233]: ACK response timeout, retry #2
[19:57:39][V][levoit:209]: Sending CMD (013140): A5.22.86.04.00.3C.01.31.40.00 (10)
[19:57:39][W][levoit:233]: ACK response timeout, retry #3
[19:57:39][V][levoit:209]: Sending CMD (013140): A5.22.87.04.00.3B.01.31.40.00 (10)
[19:57:39][W][levoit:233]: ACK response timeout, retry #4
[19:57:39][V][levoit:209]: Sending CMD (013140): A5.22.88.04.00.3A.01.31.40.00 (10)
[19:57:39][E][levoit:238]: Maximum NACKs received, skipping packet
[19:57:40][VV][scheduler:225]: Running interval 'status' with interval=5000 last_execution=865141 (now=870141)
[19:57:40][V][levoit:209]: Sending CMD (0129a1): A5.22.89.0A.00.DF.01.29.A1.00.01.7D.00.7D.00.00 (16)
[19:57:40][W][levoit:233]: ACK response timeout, retry #1
[19:57:40][V][levoit:209]: Sending CMD (0129a1): A5.22.8A.0A.00.DE.01.29.A1.00.01.7D.00.7D.00.00 (16)
[19:57:40][W][levoit:233]: ACK response timeout, retry #2
[19:57:40][V][levoit:209]: Sending CMD (0129a1): A5.22.8B.0A.00.DD.01.29.A1.00.01.7D.00.7D.00.00 (16)
[19:57:40][W][levoit:233]: ACK response timeout, retry #3
[19:57:40][V][levoit:209]: Sending CMD (0129a1): A5.22.8C.0A.00.DC.01.29.A1.00.01.7D.00.7D.00.00 (16)
[19:57:40][W][levoit:233]: ACK response timeout, retry #4
[19:57:40][V][levoit:209]: Sending CMD (0129a1): A5.22.8D.0A.00.DB.01.29.A1.00.01.7D.00.7D.00.00 (16)
[19:57:40][E][levoit:238]: Maximum NACKs received, skipping packet

@acvigue how did you find out the uart commands?

brcommander, 2025-01-07:
ok thanks. Have to deal with usb-serial adapter and tools like screen etc. But I figured out that:
1. i had to change the esp config in my yaml:
```

esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_PARTITION_TABLE_OFFSET: "0x9000"
      CONFIG_PARTITION_TABLE_CUSTOM_APP_BIN_OFFSET: "0x10000"
      CONFIG_PARTITION_TABLE_FILENAME: "partitions.csv"
```
[partitions.csv](https://github.com/user-attachments/files/18326458/partitions.csv)
2. GPIOs to be used in esphome config are: 18 (rx), 19 (tx)
3. sniffing von GPIO 19 (Testpoint 57) gave me some uart commands:
```
PowerON: A5 22 2C 07 00 B0 02 00 50 00 01 01 01
Fan Level 2: A5 22 2D 07 00 A6 02 03 55 00 01 01 02
Fan Level 1: A5 22 2E 07 00 A6 02 03 55 00 01 01 01
Fan Level 3: A5 22 2F 07 00 A3 02 03 55 00 01 01 03
Fan Level 1: A5 22 30 07 00 A4 02 03 55 00 01 01 01
Fan Level 4: A5 22 31 07 00 A0 02 03 55 00 01 01 04
PowerOFF: A5 22 33 07 00 AA 02 00 50 00 01 01 00
```
```
A5 12 F5 04 00 F8 02 00 55 00 
A5 12 F6 04 00 F7 02 00 55 00 
A5 12 F7 04 00 F6 02 00 55 00 
A5 12 F8 04 00 F5 02 00 55 00 
```
seem to be some kind of pings (every 3 seconds) / heartbeat / whatsoever

So I flashed esphome and got some log and at least there is communication:
[esphome.log](https://github.com/user-attachments/files/18326474/esphome.log)

The switches in HA for this device do not work and there are no data like pm2.5.
So I have to figure out which uart commands are used (beside the above mentioned)

swoga, 2025-01-12:
I also disassembled my Vital 100s (what a pain, with all the clips) to reverse engineer the protocol.
I sent commands via VeSync and recorded with a Logic Analyzer.
The packages are similar in structure, but not quite the same.

```
direction	action	data
MCU to ESP	status		0xA5 0x22 0x8C 0x6C 0x00 0xF7 0x02 0x00 0x55 0x00 0x00 0x01 0x01 0x01 0x03 0x05 0x00 0x01 0x02 0x01 0x00 0x03 0x01 0x00 0x04 0x01 0xFF 0x05 0x01 0x04 0x06 0x01 0x00 0x07 0x01 0x01 0x08 0x01 0x00 0x09 0x01 0x01 0x0A 0x01 0x64 0x0B 0x02 0x01 0x00 0x0E 0x01 0x00 0x0F 0x01 0x00 0x10 0x02 0x00 0x00 0x11 0x02 0x00 0x00 0x12 0x01 0x01 0x13 0x01 0x01 0x16 0x01 0x00 0x17 0x01 0x01 0x18 0x01 0x01 0x19 0x01 0x00 0x1A 0x02 0x05 0x00 0x1B 0x01 0x03 0x1C 0x01 0x01 0x1D 0x02 0x2D 0x00 0x1E 0x01 0x01 0x20 0x02 0xE0 0x01 0x1F 0x01 0x05 0x21 0x01 0x01 0x22 0x01 0x02 0x23 0x01 0x01
ESP to MCU	ack		0xA5 0x12 0x8C 0x04 0x00 0x61 0x02 0x00 0x55 0x00
MCU to ESP	status		0xA5 0x22 0x8D 0x6C 0x00 0xF6 0x02 0x00 0x55 0x00 0x00 0x01 0x01 0x01 0x03 0x05 0x00 0x01 0x02 0x01 0x00 0x03 0x01 0x00 0x04 0x01 0xFF 0x05 0x01 0x04 0x06 0x01 0x00 0x07 0x01 0x01 0x08 0x01 0x00 0x09 0x01 0x01 0x0A 0x01 0x64 0x0B 0x02 0x01 0x00 0x0E 0x01 0x00 0x0F 0x01 0x00 0x10 0x02 0x00 0x00 0x11 0x02 0x00 0x00 0x12 0x01 0x01 0x13 0x01 0x01 0x16 0x01 0x00 0x17 0x01 0x01 0x18 0x01 0x01 0x19 0x01 0x00 0x1A 0x02 0x05 0x00 0x1B 0x01 0x03 0x1C 0x01 0x01 0x1D 0x02 0x2D 0x00 0x1E 0x01 0x01 0x20 0x02 0xE0 0x01 0x1F 0x01 0x05 0x21 0x01 0x01 0x22 0x01 0x02 0x23 0x01 0x01
ESP to MCU	ack		0xA5 0x12 0x8D 0x04 0x00 0x60 0x02 0x00 0x55 0x00
ESP to MCU	power on	0xA5 0x22 0x18 0x07 0x00 0xC4 0x02 0x00 0x50 0x00 0x01 0x01 0x01
MCU to ESP	ack		0xA5 0x12 0x18 0x04 0x00 0xDA 0x02 0x00 0x50 0x00
MCU to ESP	status		0xA5 0x22 0x8E 0x6C 0x00 0xEE 0x02 0x00 0x55 0x00 0x00 0x01 0x01 0x01 0x03 0x05 0x00 0x01 0x02 0x01 0x01 0x03 0x01 0x00 0x04 0x01 0x04 0x05 0x01 0x04 0x06 0x01 0x01 0x07 0x01 0x01 0x08 0x01 0x00 0x09 0x01 0x01 0x0A 0x01 0x64 0x0B 0x02 0x01 0x00 0x0E 0x01 0x00 0x0F 0x01 0x00 0x10 0x02 0x00 0x00 0x11 0x02 0x00 0x00 0x12 0x01 0x01 0x13 0x01 0x01 0x16 0x01 0x00 0x17 0x01 0x01 0x18 0x01 0x01 0x19 0x01 0x00 0x1A 0x02 0x05 0x00 0x1B 0x01 0x03 0x1C 0x01 0x01 0x1D 0x02 0x2D 0x00 0x1E 0x01 0x01 0x20 0x02 0xE0 0x01 0x1F 0x01 0x05 0x21 0x01 0x01 0x22 0x01 0x02 0x23 0x01 0x01
ESP to MCU	ack		0xA5 0x12 0x8E 0x04 0x00 0x5F 0x02 0x00 0x55 0x00
MCU to ESP	status		0xA5 0x22 0x8F 0x6C 0x00 0xED 0x02 0x00 0x55 0x00 0x00 0x01 0x01 0x01 0x03 0x05 0x00 0x01 0x02 0x01 0x01 0x03 0x01 0x00 0x04 0x01 0x04 0x05 0x01 0x04 0x06 0x01 0x01 0x07 0x01 0x01 0x08 0x01 0x00 0x09 0x01 0x01 0x0A 0x01 0x64 0x0B 0x02 0x01 0x00 0x0E 0x01 0x00 0x0F 0x01 0x00 0x10 0x02 0x00 0x00 0x11 0x02 0x00 0x00 0x12 0x01 0x01 0x13 0x01 0x01 0x16 0x01 0x00 0x17 0x01 0x01 0x18 0x01 0x01 0x19 0x01 0x00 0x1A 0x02 0x05 0x00 0x1B 0x01 0x03 0x1C 0x01 0x01 0x1D 0x02 0x2D 0x00 0x1E 0x01 0x01 0x20 0x02 0xE0 0x01 0x1F 0x01 0x05 0x21 0x01 0x01 0x22 0x01 0x02 0x23 0x01 0x01
ESP to MCU	ack		0xA5 0x12 0x8F 0x04 0x00 0x5E 0x02 0x00 0x55 0x00
MCU to ESP	status		0xA5 0x22 0x90 0x6C 0x00 0xED 0x02 0x00 0x55 0x00 0x00 0x01 0x01 0x01 0x03 0x05 0x00 0x01 0x02 0x01 0x01 0x03 0x01 0x00 0x04 0x01 0x04 0x05 0x01 0x04 0x06 0x01 0x00 0x07 0x01 0x01 0x08 0x01 0x00 0x09 0x01 0x01 0x0A 0x01 0x64 0x0B 0x02 0x01 0x00 0x0E 0x01 0x00 0x0F 0x01 0x00 0x10 0x02 0x00 0x00 0x11 0x02 0x00 0x00 0x12 0x01 0x01 0x13 0x01 0x01 0x16 0x01 0x00 0x17 0x01 0x01 0x18 0x01 0x01 0x19 0x01 0x00 0x1A 0x02 0x05 0x00 0x1B 0x01 0x03 0x1C 0x01 0x01 0x1D 0x02 0x2D 0x00 0x1E 0x01 0x01 0x20 0x02 0xE0 0x01 0x1F 0x01 0x05 0x21 0x01 0x01 0x22 0x01 0x02 0x23 0x01 0x01
ESP to MCU	ack		0xA5 0x12 0x90 0x04 0x00 0x5D 0x02 0x00 0x55 0x00
MCU to ESP	status		0xA5 0x22 0x91 0x6C 0x00 0xEC 0x02 0x00 0x55 0x00 0x00 0x01 0x01 0x01 0x03 0x05 0x00 0x01 0x02 0x01 0x01 0x03 0x01 0x00 0x04 0x01 0x04 0x05 0x01 0x04 0x06 0x01 0x00 0x07 0x01 0x01 0x08 0x01 0x00 0x09 0x01 0x01 0x0A 0x01 0x64 0x0B 0x02 0x01 0x00 0x0E 0x01 0x00 0x0F 0x01 0x00 0x10 0x02 0x00 0x00 0x11 0x02 0x00 0x00 0x12 0x01 0x01 0x13 0x01 0x01 0x16 0x01 0x00 0x17 0x01 0x01 0x18 0x01 0x01 0x19 0x01 0x00 0x1A 0x02 0x05 0x00 0x1B 0x01 0x03 0x1C 0x01 0x01 0x1D 0x02 0x2D 0x00 0x1E 0x01 0x01 0x20 0x02 0xE0 0x01 0x1F 0x01 0x05 0x21 0x01 0x01 0x22 0x01 0x02 0x23 0x01 0x01
ESP to MCU	ack		0xA5 0x12 0x91 0x04 0x00 0x5C 0x02 0x00 0x55 0x00
ESP to MCU	fan level 3	0xA5 0x22 0x19 0x07 0x00 0xB9 0x02 0x03 0x55 0x00 0x01 0x01 0x03
MCU to ESP	ack		0xA5 0x12 0x19 0x04 0x00 0xD1 0x02 0x03 0x55 0x00
MCU to ESP	status		0xA5 0x22 0x92 0x6C 0x00 0xEC 0x02 0x00 0x55 0x00 0x00 0x01 0x01 0x01 0x03 0x05 0x00 0x01 0x02 0x01 0x01 0x03 0x01 0x00 0x04 0x01 0x03 0x05 0x01 0x03 0x06 0x01 0x01 0x07 0x01 0x01 0x08 0x01 0x00 0x09 0x01 0x01 0x0A 0x01 0x64 0x0B 0x02 0x01 0x00 0x0E 0x01 0x00 0x0F 0x01 0x00 0x10 0x02 0x00 0x00 0x11 0x02 0x00 0x00 0x12 0x01 0x01 0x13 0x01 0x01 0x16 0x01 0x00 0x17 0x01 0x01 0x18 0x01 0x01 0x19 0x01 0x00 0x1A 0x02 0x05 0x00 0x1B 0x01 0x03 0x1C 0x01 0x01 0x1D 0x02 0x2D 0x00 0x1E 0x01 0x01 0x20 0x02 0xE0 0x01 0x1F 0x01 0x05 0x21 0x01 0x01 0x22 0x01 0x02 0x23 0x01 0x01
ESP to MCU	ack		0xA5 0x12 0x92 0x04 0x00 0x5B 0x02 0x00 0x55 0x00
MCU to ESP	status		0xA5 0x22 0x93 0x6C 0x00 0xEB 0x02 0x00 0x55 0x00 0x00 0x01 0x01 0x01 0x03 0x05 0x00 0x01 0x02 0x01 0x01 0x03 0x01 0x00 0x04 0x01 0x03 0x05 0x01 0x03 0x06 0x01 0x01 0x07 0x01 0x01 0x08 0x01 0x00 0x09 0x01 0x01 0x0A 0x01 0x64 0x0B 0x02 0x01 0x00 0x0E 0x01 0x00 0x0F 0x01 0x00 0x10 0x02 0x00 0x00 0x11 0x02 0x00 0x00 0x12 0x01 0x01 0x13 0 [... cut ...]
MCU to ESP	status		0xA5 0x22 0x9A 0x6C 0x00 0xEC 0x02 0x00 0x55 0x00 0x00 0x01 0x01 0x01 0x03 0x05 0x00 0x01 0x02 0x01 0x00 0x03 0x01 0x00 0x04 0x01 0xFF 0x05 0x01 0x01 0x06 0x01 0x00 0x07 0x01 0x01 0x08 0x01 0x00 0x09 0x01 0x01 0x0A 0x01 0x64 0x0B 0x02 0x01 0x00 0x0E 0x01 0x00 0x0F 0x01 0x00 0x10 0x02 0x00 0x00 0x11 0x02 0x00 0x00 0x12 0x01 0x01 0x13 0x01 0x01 0x16 0x01 0x00 0x17 0x01 0x01 0x18 0x01 0x01 0x19 0x01 0x00 0x1A 0x02 0x05 0x00 0x1B 0x01 0x03 0x1C 0x01 0x01 0x1D 0x02 0x2D 0x00 0x1E 0x01 0x01 0x20 0x02 0xE0 0x01 0x1F 0x01 0x05 0x21 0x01 0x01 0x22 0x01 0x02 0x23 0x01 0x01
ESP to MCU	ack		0xA5 0x12 0x9A 0x04 0x00 0x53 0x02 0x00 0x55 0x00

```

swoga, 2025-01-12:
The values in the status update look like tag-length-value pairs.
I have marked all those that fit into the scheme with square brackets:

```
head type seq  len  ?    chk  ?    type type ?    ?    ?    ?    ?    ?    ?    ?    TLV pairs...
0xA5 0x22 0x93 0x6C 0x00 0xEB 0x02 0x00 0x55 0x00 0x00 0x01 0x01 0x01 0x03 0x05 0x00 [ 0x01 0x02 0x01 0x01 ] [ 0x03 0x01 0x00 ] [ 0x04 0x01 0x03 ] [ 0x05 0x01 0x03 ] [ 0x06 0x01 0x01 ] [ 0x07 0x01 0x01 ] [ 0x08 0x01 0x00 ] [ 0x09 0x01 0x01 ] [ 0x0A 0x01 0x64 ] [ 0x0B 0x02 0x01 0x00 ] [ 0x0E 0x01 0x00 ] [ 0x0F 0x01 0x00 ] [ 0x10 0x02 0x00 0x00 ] [ 0x11 0x02 0x00 0x00 ] [ 0x12 0x01 0x01 ] [ 0x13 0x01 0x01 ] [ 0x16 0x01 0x00 ] [ 0x17 0x01 0x01 ] [ 0x18 0x01 0x01 ] [ 0x19 0x01 0x00 ] [ 0x1A 0x02 0x05 0x00 ] [ 0x1B 0x01 0x03 ] [ 0x1C 0x01 0x01 ] [ 0x1D 0x02 0x2D 0x00 ] [ 0x1E 0x01 0x01 ] [ 0x20 0x02 0xE0 0x01 ] [ 0x1F 0x01 0x05 ] [ 0x21 0x01 0x01 ] [ 0x22 0x01 0x02 ] [ 0x23 0x01 0x01 ]
```

For example `[ 0x04 0x01 0x03 ]` is tag `0x04`, which should be fan speed, length is 1 and value is 3.

SomeFixItDude, 2025-01-14:
I saw there is a pull request to support the 200S.  Does this work yet?  I would love to flash mine but I have to have a high WAF factor :laughing:

rwtallant13, 2025-01-16:
Just use the forked version that the pull request is from for the 200s. Working perfectly for me

https://github.com/Havarius/core200s/tree/core200s-support

SomeFixItDude, 2025-01-16:
@rwtallant13 OK I am going for it.  Where did you take apart the control panel?  Or any advice on the disassembly?

richtiger, 2025-01-16:
> [@rwtallant13](https://github.com/rwtallant13) OK I am going for it. Where did you take apart the control panel? Or any advice on the disassembly?
> 
> I undid the bottom, took filter out, removed the 4 screws and separated the bottom. This leaves the black motor housing. When I try to lift it out it is resistant and I don't want to damage the control. Did you just tug it the rest of the way out?
> 
> Thanks again
I am using the code from that fork as well without issues.

The top just snaps in, you can push it down through the inside, once its out you can disconnect the board and remove the 3 screws and the top board will come out.

SomeFixItDude, 2025-01-16:
Should I be continuing in this bottom direction or trying to pry something from the top?

[image]

Thanks again for the help

SomeFixItDude, 2025-01-16:
oh sorry, was too fast, I pushed down like you said and tada.  thanks again!

SomeFixItDude, 2025-01-16:
Just having one problem, looking for help.  I changed the esphome yaml to:

external_components:
  - source: github://Havarius/core200s/tree/core200s-support
    components: [levoit]

levoit:
  id: levoit1
  model: core200s

but it complains:
Unknown value 'core200s', did you mean 'core400s', 'core300s'?.

Maybe I am not doing the external_components section correctly?

Thanks Again

SomeFixItDude, 2025-01-16:
Sorry for all the posts, I have figured out how to add the component.  Seems to crash often though :(

Uptime:

[image]

richtiger, 2025-01-17:
> Just having one problem, looking for help. I changed the esphome yaml to:
>
> external_components:
>
> * source: github://Havarius/core200s/tree/core200s-support
>   components: [levoit]
>
> levoit: id: [...]

You have to add the branch:
source: github://Havarius/core200s@core200s-support

SomeFixItDude, 2025-01-17:
I made just a few small tweaks to that fork and I am no longer crashing/rebooting.  Also changed the heartbeat and a status update after someone changes a switch or fan so the value is updated right away.  This way you don't see the switch go on and then off and then back to on again.  I don't use github very much, what is the proper etiquette, do I send a pull request to the repo with the core200s-support or do I put it here?  Really didn't do that much changes but it has made my 200s work really well.

rwtallant13, 2025-01-18:
I personally might submit it to the other fork since they seem a bit more responsive and then wait for that fork to get merged here as well. 

Either way I would be interested in what tweaks you made too. Mine works well enough but am curious.

SomeFixItDude, 2025-01-28:
@rwtallant13  sorry for delay, I added my code as a pull request.  I still have some remaining features to work out, such as the timer but here is a nice first go at it for me.

SomeFixItDude, 2025-01-28:
Ahh I added the heart beat thing last minute and it seems to have caused some instability.  I am going to cancel the pull request and track down the issue.  Shouldn't have jumped the gun.

SomeFixItDude, 2025-01-28:
Ok forgot to mention added configurable timeout and delay, and fix for the PM2.5 sensor as well.  Which works for the 200S.  Not sure if someone can try it out for the other models.  I only have the 200S.

SomeFixItDude, 2025-01-29:
Just released a bunch of fixes on my fork.  Does anyone know how to reset the filter light?  Currently working on timers and display brightness slider.  The command I have seems to make it blink every status.  but on restart of the esp32 it clears and stays off.

wpmjones, 2025-02-08:
Looking for some clarity.  I just received my Vital 200s (not a Core 200s) and want to get this set up on esphome.  @SomeFixItDude are you working with a Vital series?  I noticed that your fork still says Core, so I wanted to be sure.

SomeFixItDude, 2025-02-09:
@wpmjones Sorry, I saw the pull request for 200 support and this thread and thought it was about the 200s.  If I had a read a little closer I would have seen it was about a different model.  My fork just cleaned up the communications (timeouts, errors, and at least my esp was crash rebooting), pm2.5, air quality index, added a filter reset button.  Sorry!  I would help out if I had a vital to test with.  However, I could help out with adjusting the code to rx and tx different payloads though.

GO BIRDS!

Scags104, 2025-03-02:
@SomeFixItDude @wpmjones  Would love to see more work on the Vital200... I have one and an analyzer. Not sure what to do with it beyond connecting pins though....

Scags104, 2025-03-03:
> > [@SomeFixItDude](https://github.com/SomeFixItDude) [@wpmjones](https://github.com/wpmjones) Would love to see more work on the Vital200... I have one and an analyzer. Not sure what to do with it beyond connecting pins though....
> 
> [@Scags104](https://github.com/Scags104) I'll fork another version and update the code to support different payload styles based on the model. I also plan to clean up the code since my initial version was put together quickly. My goal is to get it working with my Core200 first. After that, it should be much easier for others to add support for their models without digging through the code—everything will be more centralized.
> 
> Right now, I’m in the middle of transitioning my hypervisor from XCP-ng to Proxmox, so I’ll likely be a couple of weekends out.

let me know when youre ready! Thanks

glitchykitten, 2025-09-07:
Does anyone have a guess if the 600s would flash and work as expected?

SomeFixItDude, 2025-09-07:
Sorry priorities have shifted I'll try to come back here when I can. IDK if anyone else can take a swing at it. @glitchykitten  you could always backup the current firmware and give it a shot.  Then if it doesn't work restore back.

glitchykitten, 2025-11-22:
I foolishly set up the VeSync web support and never got back to hacking on this.

targor, 2025-11-28:
As for the Vital 200s, it now has its own component, see: https://github.com/acvigue/esphome-levoit-air-purifier/issues/36#issuecomment-3590688401