acvigue/esphome-levoit-air-purifier #6: Issues with Core 200s

jeffreygarc, 2024-06-12:
Hello!

I installed this on my Core 200s and am having a few issues. The esp keeps rebooting and is only online for a couple of seconds.

I get these errors in the logs:
```
[16:18:18][E][levoit:078]: Received error response, ignoring packet
[16:18:18][W][levoit:233]: ACK response timeout, retry #1
[16:18:19][E][levoit:078]: Received error response, ignoring packet
[16:18:19][W][levoit:233]: ACK response timeout, retry #2
[16:18:19][E][levoit:078]: Received error response, ignoring packet
[16:18:19][W][levoit:233]: ACK response timeout, retry #3
[16:18:19][E][levoit:078]: Received error response, ignoring packet
[16:18:19][W][levoit:233]: ACK response timeout, retry #4
[16:18:19][E][levoit:078]: Received error response, ignoring packet
[16:18:19][E][levoit:238]: Maximum NACKs received, skipping packet
```

I have also noticed that no matter what speed I change the fan to in HA, the fan speed is always set to three with this in the logs:
```
[16:47:02][D][fan:021]: 'Levoit Purifier' - Setting:
[16:47:02][D][fan:024]:   State: ON
[16:47:02][D][fan:030]:   Speed: 3
```

Here is my esphome config
```
esphome:
  name: main-bedroom-air-purifier
  friendly_name: Main-bedroom-air-purifier

esp32:
  board: esp32dev
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_FREERTOS_UNICORE: y

external_components:
  - source: github://acvigue/esphome-levoit-purifier
    components: [levoit]

# Enable logging
logger:
  baud_rate: 115200
  level: VERBOSE

# Enable Home Assistant API
api:
  encryption:
    key: "..."

ota:
  password: "..."

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Main-Bedroom-Air-Purifier"
    password: "..."
captive_portal:

uart:
  tx_pin: GPIO17
  rx_pin: GPIO16
  baud_rate: 115200

levoit:
  id: levoit1

fan:
  - platform: levoit
    name: Levoit Purifier

sensor:
  # Uptime sensor.
  - platform: uptime
    name: Uptime

  # WiFi Signal sensor.
  - platform: wifi_signal
    name: WiFi Signal
    update_interval: 60s

switch:
  - platform: levoit
    display_lock:
      name: "Display Lock"
    display_on:
      name: "Display On"
    power:
      name: "Master Power"

select:
  - platform: levoit
    fan_mode:
      name: "Fan Mode"
```

Thanks!

acvigue, 2024-06-14:
Please try
```yml
levoit:
  id: levoit1
  model: core200s #this line
```

closed 2024-07-21 by acvigue

jeffreygarc, 2024-08-06:
Sorry for not replying, I have not had time to get back to this.

I added the line in my config, but ESPHome doesn't compile:

```
INFO ESPHome 2024.7.3
INFO Reading configuration /config/esphome/levoit.yaml...
Failed config

levoit: [source /config/esphome/levoit.yaml:44]
  id: levoit1
  
  Unknown value 'core200s', did you mean 'core400s', 'core300s'?.
  model: core200s
```

Thanks!

rwtallant13, 2024-12-12:
Sorry to reopen this. The air purifier works and connects to esphome but I'm having this same issue. I tried your given solution but I'm getting this:
```

INFO ESPHome 2024.11.3
INFO Reading configuration /config/esphome/levoit200s.yaml...
Failed config

levoit: [source /config/esphome/levoit200s.yaml:46]
  id: levoit1
  
  Unknown value 'core200s', did you mean 'core400s', 'core300s'?.
  model: core200s
```

Am I doing something wrong in the config or is this unneeded?