acvigue/esphome-levoit-air-purifier #7: Missing documentation for flashing

Majr25, 2024-07-20:
Documentation should mention you need to wire the IO0 pin to ground in order to flash the ESP32 chip.

Californian, 2024-07-24:
Wow thanks for this, unfortunately it still doesn't work to flash it for me -- any other things you found missing in the instructions?

Majr25, 2024-07-25:
Nothing else I can think of for actually doing the flash, if you're able to read the contents fine so you know you've got the rx/tx wired up correctly. If you do the flash but get errors when booting (after disconnecting IO0), make sure you picked the modern image when downloading from esphome.

Californian, 2024-07-26:
Unfortunately, `sudo esptool.py chip_id` just fails, so I assume I wired something else up incorrectly (though it looks like nothing is shorted based on my continuity tests). Thanks for solving at least one of my issues though!

```
Serial port /dev/ttyUSB0
Connecting......................................
/dev/ttyUSB0 failed to connect: Failed to connect to Espressif device: No serial data received.
```

OutsmartinTech, 2025-03-04:
Just a little note for anyone stuck on this in the future, I had to wire IO0 to ground, as indicated above. I also had to swap TX/RX pins (TX on board connected to RX on the UART converter and RX on board connected to TX on the UART converter). I'm not sure if that's fairly standard, but may trip up a newbie like me.

ewrogers, 2025-05-03:
+1 on grounding `IO0` pin to read/flash firmware.

SeveranExp, 2025-10-17:
Strange, yeah they are gone. I’ve asked one of the mods if they can help. If not I’ll try to find them in my Google Photos folder,  no worries .