acvigue/esphome-levoit-air-purifier #8: Can you please share raw protocol notes

kquinsland, 2024-08-04:
Hi!

I am attempting to reverse engineer one of their humidifiers.
It has the same overall architecture: ESP32 connects to the cloud and talks to the main MCU over a 9600 baud UART.

I am seeing packets that are _similar_ but not quite the same.
E.g:

```
a5 22 01 04 00 e1 01 11 40 00
```

I can read through the c++ code to see how you validate/parse each message but I was hoping you'd have some basic "plain text" docs / notes you can share?

I dumped the firmware and was very happy to see that they put the function name as an arg to all of their `printf()` logging so that's plan B.

acvigue, 2024-08-08:
Hi it seems to be the same format with an exchanged byte of e1 instead of the second 00.

General fmt:
A5 - static
message type (ack, nack, cmd?)
sequence number (increase by one every sent msg)
payload length (plus 3, payload starts after the three byte payload type) i.e payload of [0x00] has PL of 4
empty byte - 0x00
message checksum - see lines 197-202 in levoit.cpp, is calculated AFTER entire message is assembled
payload type pt 1
payload type pt 2
payload type pt 3

...payload...

kquinsland, 2024-08-08:
> Hi it seems to be the same format

Yeah, the "outer" format is more or less the same but in addition to basic request/ack packets, I'm seeing messages of type `0x02` which always have a sequence of `0x00` and payload length of 19 bytes!

As best I can tell, this is the MCU sending a comprehensive "here's the status of every device i monitor/control" message.

This device also has a fan and a wifi indicator light but the payloads you've discovered for controlling the fan ... etc don't seem to match what I'm seeing.

closed 2024-08-08 by kquinsland