acvigue/esphome-levoit-air-purifier #9: No value for PM Sensor

Zerogiven, 2024-10-24:
Hi there,

First thanks a lot for doing this great work, it was one reason why i've choose the 300s purifier.
Flashing worked like a charm and i already linked it to my HA instance.

The only "problem" i have, but maybe this is an expected behavior, i only get a value for the PM Sensor if i turn on the fan with manual or auto mode. So why is this a problem for me: Cause i want to turn the purifier on if the PM Sensor is above some value, otherwise i would like to put it into sleep mode. Another option would be that you can set it to 0 at manual mode (not working, 0-1 is the same fan speed at manual mode)

If it's an expected behavior, maybe you have an idea how i can "hack" it to get the PM sensor values also if the fan is in sleep mode.

I think the fan sleep mode is using the deep sleep from the esp32. Not sure, however, maybe we could make another mode where the esp32 is still running, but the fan has just 1% power (or something above where you still don't hear the fan).

cheers
Christopher

hbrennhaeuser, 2025-02-05:
I'm an issue that may be related to yours, see #18, for me it helped to use the older commit df394f8.

Using this version pm_2_5 is updated when in sleep mode and even with "Master Power" turned off, at least on my device.

Maybe this fixes your issue as a temporary workaround.

```yaml
external_components:
  - source: github://acvigue/esphome-levoit-purifier@df394f8bf93671446e072264791f81c973fa8dec
    components: [levoit]
```

jojo01fr, 2025-02-14:
> I'm having an issue that may be related to yours, see [#18](https://github.com/acvigue/esphome-levoit-air-purifier/issues/18), for me it helped to use the older commit [df394f8](https://github.com/acvigue/esphome-levoit-air-purifier/commit/df394f8bf93671446e072264791f81c973fa8dec).
> 
> Using this version pm_2_5 is updated when in sleep mode and even with "Master Power" turned off, at least on my device.
> 
> Maybe this fixes your issue as a temporary workaround.
> 
> external_components:
>   - source: github://acvigue/esphome-levoit-purifier@df394f8bf93671446e072264791f81c973fa8dec
>     components: [levoit]

Hi this solves the PM2.5 issue but refuses the Filter Reset Button, need to comment that part to be able to compile.

traxeon, 2025-02-18:
I'm also having the same issue.  No PM sensor and switching to the other code base results in a filter reset error.

robert-alfaro, 2025-02-21:
Correct me if I'm wrong, but the Filter Reset is kind of useless since the Filter Reset Indicator is normally illuminated by the cloud. IOW the levoit device is not tracking filter life so the Filter Reset is useless to date.

Zerogiven, 2025-04-20:
The filter reset could be simulated or made with another esphome component i think 🤔

PR #26 (2025-10-06): Fix PM2.5 values always reporting as 0
This patch corrects the PM2.5 value scaling logic for 300S/400S

Previously, the raw PM2.5 value was scaled using `(value * 10) / 10000`, which caused almost all real-world readings to truncate to zero due to integer division. As a result, Home Assistant always displayed 0.0 for PM2.5.

This fix removes the unnecessary division and scales the raw value with `(value * 10)` to align with the expected internal format used by the existing PM2.5 sensor logic.

The resulting values now match the 400S display and report correctly in Home Assistant. This was also [... cut]