acvigue/esphome-levoit-air-purifier #11: Adding Levoit Core 200s

richtiger, 2024-12-04:
Has anyone done anything with the 200s lately?

I was able to get the 200s working by duplicating what was added for the 400 with a couple differences. However, I can’t seem to figure out the nightlight, the AQI value is 0 and the PM25 value is 65000. 

I put factory firmware back on, connected it to the app and issued commands from the app while my logic analyzer is connected, but not getting anything. Connectly both tx and rx to its own channel (using Logic, set as async serial with baud at 115200) and should see some, right?

Comparing the 200 board to 300, it uses 5v instead of 3.3v but can’t tell of any other differences. I do see pads for grnd, 5v, clk and dio - should I be using those?

rwtallant13, 2024-12-13:
Would you mind sharing your yaml config? I just did this to my 200s with the default 300s config and would like to have some more control. Thanks.

richtiger, 2024-12-31:
I've been using the updates from https://github.com/acvigue/esphome-levoit-air-purifier/pull/13 and everything has been working with a 200s.

closed 2024-12-31 by richtiger

acvigue, 2025-01-19:
#13 has been merged.