acvigue/esphome-levoit-air-purifier #15: Lots of Ack failures in logs

haydonryan, 2025-01-19:
Hi Aiden,
On the latest esphome i'm getting a ton of ack response timeouts.  The logs just keep flowing with these errors.

```
[20:56:37][C][levoit:066]: Levoit!
[20:56:38][W][levoit:233]: ACK response timeout, retry #1
[20:56:38][W][levoit:233]: ACK response timeout, retry #2
[20:56:38][W][levoit:233]: ACK response timeout, retry #3
[20:56:38][W][levoit:233]: ACK response timeout, retry #4
[20:56:38][E][levoit:238]: Maximum NACKs received, skipping packet
[20:56:39][W][levoit:233]: ACK response timeout, retry #1
[20:56:39][W][levoit:233]: ACK response timeout, retry #2
[20:56:39][W][levoit:233]: ACK response timeout, retry #3
[20:56:39][W][levoit:233]: ACK response timeout, retry #4
[20:56:39][E][levoit:238]: Maximum NACKs received, skipping packet
```

acvigue, 2025-01-19:
Hi! not seeing this on my 300s

acvigue, 2025-01-21:
I did just update the 300s I have at college and the other 300s I have back home and still not able to replicate this unfortunately.

closed 2025-02-03 by acvigue

stevenrhawk, 2025-08-06:
I'm seeing persistent ACK errors as well. I believe the Core 300s may have another hardware revision with different UART communication happening. I tried sniffing packets on UART and was able to come up with a few things, but I'll be honest I was way out of my depth. I've got the factory firmware bin dumped if anyone has any interest in it.

acvigue, 2025-08-09:
Can you upload logs of your packet capture? Would be helpful to have a bit of "inactivity" where it is just sending things on its own, then record, press something on app, tag, and repeat for a few more.