acvigue/esphome-levoit-air-purifier #20: ACK Timeouts

Grafton1046, 2025-03-05:
I am no longer able to properly control the 300s, commands from esphome attempt and then fail back to their previous state.

it's a similar issue to https://github.com/acvigue/esphome-levoit-air-purifier/issues/15

[logs_airpurifier_logs.txt](https://github.com/user-attachments/files/19089442/logs_airpurifier_logs.txt)

Grafton1046, 2025-03-08:
Update, rolled ESPhome back to 2024.12.4 still seeing the same timeout issues

Grafton1046, 2025-03-09:
Update to my update 
Rolled back to esphome 2024.12.2 still seeing ack issues

Rolled the repository back to the 12/30/24 commit, was still seeing ack issue, until I power cycled the entire air purifier. Will test with everything updated.

jojo01fr, 2025-04-19:
Same here, obliged to power cycle the device but ack timeouts are back a few days later. Will have to put a smartplug with some automation on it if no solution is found.