acvigue/esphome-levoit-air-purifier #21: Levoit Core 200S - Fan Speed control

krahsdevil, 2025-03-28:
Hi.
I've succesfully installed it on a Levoit Core 200S and all seems to be working fine however, I don't have the option to control the fan speed (0, 1, 2 or 3) in HA (in local web server is possible, but there is no an entity for it on HA, only one called "Levoit Purifier" which is a switch, where you can stop or initiate the fan).
Having this I think the implementation would 100% perfect :-).

Thanks! 
Regards.

richtiger, 2025-03-28:
Levoit Purifier is a fan, click on the icon and it should give you speed options.

krahsdevil, 2025-03-28:
Oh… you are right :-S I didn’t realize on that!
Thank you, it works fine.

kerkmann, 2025-05-17:
@krahsdevil can this be closed, otherwise people could be confused regarding the issue title? :)

closed 2025-05-17 by krahsdevil