acvigue/esphome-levoit-air-purifier #28: Levoit Core 200s is reporting a wrong pm2.5 value

kerkmann, 2025-05-15:
I have five Levoit core 200s and all of them are reporting just `0` or `6.4`, I wasn't able to figure out what the root cause is. I have the original firmware, maybe it helps out to investage there as well. :)

neilmartin83, 2025-05-15:
I think the Core 200s does not have a PM2.5 sensor. I couldn't see anything in the PCB and when I flashed mine and there's no mention in the product specs etc. So I just removed the block for it in the ESPHome config.

kerkmann, 2025-05-16:
> I think the Core 200s does not have a PM2.5 sensor. I couldn't see anything in the PCB and when I flashed mine and there's no mention in the product specs etc. So I just removed the block for it in the ESPHome config.

Thank you very much! That makes sense, I was just using the [sample-levoit-200s-configuration.yaml](https://github.com/acvigue/esphome-levoit-air-purifier/blob/main/sample-levoit-200s-configuration%20.yaml#L87). ^^"
The `Core 200s` has then also no `air quality index`, am I right? :)

neilmartin83, 2025-06-01:
Yeah - I believe so. I was hoping Levoit might have included the sensors but just disabled them in the stock firmware etc but no such luck! The 200S is just a simple fan/night light basically.

PR #29 (2025-10-02): Fix wrong documentation core 200s
Fix some misleading documentation. :)

Closes #28
commit a6b880b3 (2025-10-02): fix: remove sensor section from sample-levoit-200s-configuration.yaml

The `levoit core 200s` has no `pm2.5` sensor.

Closes #28
closed 2025-10-02 by acvigue