acvigue/esphome-levoit-air-purifier #32: Cannot compile the firmware: Error while finding module

bangelischan, 2025-07-07:
ESPHome Builder version: 2025.6.3

Yaml: 

`esphome:
  name: "levoit-air-purifier"
  friendly_name: "levoit-air-purifier"

esp32:
  board: esp32dev
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_FREERTOS_UNICORE: y

external_components:
  - source: github://acvigue/esphome-levoit-purifier
    components: [levoit]

logger:
  baud_rate: 115200

# Enable Home Assistant API
api:

ota:
  - platform: esphome

wifi:
  networks:
    - ssid: !secret wifi_ssid
      password: !secret wifi_password

  ap:
    ssid: "Levoit-Core300S Fallback Hotspot"
    password: !secret ap_password

captive_portal:

uart:
  tx_pin: GPIO17
  rx_pin: GPIO16
  baud_rate: 115200

levoit:
  id: levoit1
#  model: core300s  # default core300s
#  command_delay: 300 # default 300
#  command_timeout: 200 # default 200
#  status_poll_seconds: 0 # default 0, not needed if you get auto status msgs

fan:
  - platform: levoit
    name: Levoit Purifier

  # PM2.5 sensor
sensor:
  - platform: levoit
    pm_2_5:
      name: "PM2.5"
    air_quality:
      name: "Air Quality Index"
switch:
  - platform: levoit
    display_lock:
      name: "Display Lock"
    display_on:
      name: "Display On"
    power:
      name: "Master Power"

button:
  - platform: levoit
    filter_reset:
      name: "Filter Reset"

select:
  - platform: levoit
    fan_mode:
      name: "Fan Mode"
    auto_mode:
      name: "Auto Mode"
`
Error: 
`INFO ESPHome 2025.6.3
INFO Reading configuration /config/esphome/tests.yaml...
WARNING Using `fan.FAN_SCHEMA` is deprecated and will be removed in ESPHome 2025.11.0. Please use `fan.fan_schema(...)` instead. If you are seeing this, report an issue to the external_component author and ask them to update it. https://developers.esphome.io/blog/2025/05/14/_schema-deprecations/. Component using this schema: fan
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-protocols.git@mdns-v1.8.2
INFO Compiling app...
Processing levoit-air-purifier (board: esp32dev; framework: espidf; platform: https://github.com/pioarduino/platform-espressif32/releases/download/53.03.13/platform-espressif32.zip)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.50302.0 (5.3.2) 
 - tool-cmake @ 3.30.2 
 - tool-esptoolpy @ 4.8.6 
 - tool-mklittlefs @ 3.2.0 
 - tool-ninja @ 1.7.1 
 - tool-riscv32-esp-elf-gdb @ 14.2.0+20240403 
 - tool-xtensa-esp-elf-gdb @ 14.2.0+20240403 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp-elf @ 13.2.0+20240530
Reading CMake configuration...
-- git rev-parse returned 'fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).'
-- Building ESP-IDF components for target esp32
-- Configuring incomplete, errors occurred!

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:552 (message):
  /root/.platformio/penv/.espidf-5.3.2/bin/python: Error while finding module
  specification for 'idf_component_manager.prepare_components'
  (ModuleNotFoundError: No module named 'idf_component_manager')

Call Stack (most recent call first):
  /data/cache/platformio/packages/framework-espidf/tools/cmake/project.cmake:710 (idf_build_process)
  CMakeLists.txt:3 (project)

========================== [FAILED] Took 1.00 seconds ==========================`

Looks like a version issue 👀

acvigue, 2025-10-07:
Can't reproduce this on latest builder, would you mind trying again? Feel free to reopen or ping if it still occurs.

closed 2025-10-07 by acvigue

bugbountyguy, 2026-01-01:
I'm running into this issue as well. esphome 2025.12.2

```

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /config/.esphome/platformio/packages/framework-espidf/tools/cmake/build.cmake:629 (message):
  /root/.platformio/penv/.espidf-5.5.1/bin/python: Error while finding module
  specification for 'idf_component_manager.prepare_components'
  (ModuleNotFoundError: No module named 'idf_component_manager')

Call Stack (most recent call first):
  /config/.esphome/platformio/packages/framework-espidf/tools/cmake/project.cmake:740 (idf_build_process)
  CMakeLists.txt:3 (project)
```
What was the resolution to this?