acvigue/esphome-levoit-air-purifier #33: Question: Push code upstream?

DarkSpir, 2025-07-17:
Hey there,

thank you for your work here, this is a cool project. One question from my side: Since your fork is now almost 10.000 commits behind the upstream repository, do you plan to push the levoit code to the esphome project?

acvigue, 2025-07-19:
Hi! not really as this is just a custom component, we don't hold the entire source of ESPHome, and as such, don't need to keep it up to date. Not planning to create a PR to merge it into ESPHome, as we're basically "hijacking" a commercial product.

vasileio, 2025-07-20:
Just chiming in as my PR for https://devices.esphome.io/ has just been merged: https://github.com/esphome/esphome-devices/pull/1157

I've credited @acvigue and this will at least expose his great work to a wider audience of like-minded folks, I hope. I thoroughly enjoyed making this device *actually* my own, so that was my way of giving back!

acvigue, 2025-07-27:
That's great it fully works. I'll go ahead and update the readme.